IEEERobotics/bot #393: Kinnect based computer vision

SeanKetring, 2015-09-16:
Task: 

Investigating the kinnect as a possible solution for computer vision.  Locating possible sources for cheap hardware, and information on hacking and implementing the hardware.

Deliverable:

Recording the findings of your research on this issue, and reporting them to the team next all hands meeting (9/15)

AhmedSamara, 2015-09-16:
(Issue updated, please click through to view on github)

tabarrett, 2015-09-16:
Do you just comment to become a Participant/Assignee or do we need to be assigned?

SeanKetring, 2015-09-16:
If your interested in doing some research on this topic, feel free to self assign, we are not going to assign tasks to people, because we want people to choose their own so that they work on what they are most interested in

jwschaff, 2015-09-17:
I'll help out on this one.

LaurenBrennan, 2015-09-17:
As will I

shreyashgotee, 2015-09-17:
I wanna be a part of this too!!

originaldumbbaby, 2015-09-17:
some of what i've been reading suggests monte carlo localization would work well with the kinect, especially since we already know the course dimensions. anyone have experience with this?

RyanQuinnNCSU, 2015-09-18:
So things I found on Hack-day. The kinetics dimensions is 11" x 2.5" x 1.25", so I think it meets the size requirements. The general consistence on the internet for the weight of the Kinetic is 3.1 pounds.  As far as tech support for using the Kinect, the best info I found was in this link "https://learn.adafruit.com/hacking-the-kinect/overview" . For pricing, the cheapest Kinect I found was a $26 Microsoft Xbox 360 Kinect Sensor with shipping include (google shopping).

tabarrett, 2015-09-22:
I found this proof of concept video:  and also this site which is probably more helpful: http://www.intorobotics.com/working-with-kinect-3d-sensor-in-robotics-setup-tutorials-applications/ 
It has a series of guides that cover how to set up the kinect, how use specific features of the kinect. an Example is this link from the guide that has example solid work files for building the mount: http://www.hsi.gatech.edu/hrl-wiki/index.php/Kinect_Sensor_Mount. It also has several guide on 3-d mapping with kinect, detecting distance from object and I think one that explains how to detect color.

kakeim, 2015-09-22:
I plan to help on this.

jmberman, 2015-09-22:
I would like to help on this too.

Nahc7893, 2015-09-23:
I would like to help with this.

SeanKetring, 2015-09-24:
The kinnect 1 has a strange adapter which would make interfacing it with our beagle bone complicated,

We found the following adapter, 
http://www.amazon.com/Kinect-Adapter-Windows-xbox-one/dp/B00NMSHT7E

which looks like it might work.  We need some info  on the power requirements of the Kinnect, (most likely runs of low voltage DC).  Can someone take on figuring out how we could use the linked adapter unit to power the Kinnect 1 by our bot?

ZiadAli, 2015-09-25:
I remembered something about the Kinect today which I am worried may be slightly problematic. The Kinect works best at a distance of greater than 2 feet away from its target, meaning that it can't accurately measure distances closer than that. It still functions fine as a camera, so its QR-code reading ability wouldn't be impaired, but wouldn't this throw a wrench in our plans to use the Kinect?

AhmedSamara, 2015-09-25:
That's a great point ziad.

The QR code reading is a part of the senior design project, and it will
likely be done from a separate camera mounted on the arm itself (to allow
for corrections in the arms position as it picks up blocks) so that part
doesn't change, but the fact that it works best at farther ranges throws a
huge wrench in our hopes of using it on the course.

Not necessarily for QR codes, but the course in general.

jwschaff, 2015-09-25:
That's actually been known to some of us that were working on this before. There is a solution though. There is a lens that can be put on the Kinect that moves the operational range closer to the sensor. One of the team members has a Kinect and the lens together he is willing to donate to the team, so we can still continue exploring this. There may still end up being issues with distance, but its not worth taking off the table yet.

ZiadAli, 2015-09-25:
I'm familiar with the lens, but my understanding is that it only decreases the operational range to something like a 1.25-1.5 ft, which still makes me skeptical about the Kinect's usability. However, if there is a lens that works better, or if that would be sufficient visibility, then by all means go for it. I just don't want to see a lot of people invest time in the Kinect if its going to be a dead end.

Nahc7893, 2015-09-29:
https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=5&cad=rja&uact=8&sqi=2&ved=0CCoQFjAEahUKEwjMj_yesZ3IAhWIy4AKHTuOBUg&url=http%3A%2F%2Fwww.cs.cornell.edu%2Fcourses%2Fcs7670%2F2011fa%2Flectures%2Fzhaoyin_kinect.pdf&usg=AFQjCNGSfm9RiH4O2NNjIfJf4PRLQf3TaA&sig2=TFGCtg0rgVU3XdmYTIOt5A&bvm=bv.103627116,d.eXY

Sidd-GrizzAp3, 2015-09-30:
guys check this out: 

http://wiki.ros.org/kinect AWESOME RESOURCE.

Sidd-GrizzAp3, 2015-09-30:
guys please check this out: 
#408

relnor, 2015-10-24:
Other Kinect like sensors that are not kinect.

asus xtion pro live 
http://www.amazon.com/Asus-Xtion-PRO-LIVE-90IW0122-B01UA/dp/B00CET0ZBO
Pros
runs solely on usb power, openni support
cons
not sure about freenect support. seems to be some support for depth imagings

Apple Primesense Carmine 1.09 - 3D Webcam Sensor
http://www.amazon.com/dp/B00KO908MM?tag=tellgadget-20
pros
Basically the guys who made the first kinect got bought out by apple and produced another kinect.
Openni support
From what I see it is only usb power 
Available at the dh hill makerspace

cons
not sure about freeenect support

relnor, 2015-10-24:
Also There is a kinect v2 for windows at the makerspace.

relnor, 2015-10-24:
It appears that primasense is usb powered
http://wiki.ipisoft.com/Depth_Sensors_Comparison

relnor, 2015-10-26:
The kinect v2 requires usb 3.0 ports. I'm pretty sure beaglebone only has version 2.0

tabarrett, 2015-10-28:
So is the Kinect finished as a Localization method? Or are we still considering it?

jwschaff, 2015-10-28:
From what research I have done both online and with the hardware itself, the kinect is definitely a viable method of localization. It is capable of both identifying objects and returning their distance at at usable range. I suppose it comes down to weither or not we actually want to invest the time into creating support for it as a localization method.