IEEERobotics/bot #394: Drivable Chassis

SeanKetring, 2015-09-16:
Task:

Building a lower bot to competition specs that has all the main components necessary in order to take in user commands and respond accordingly.   It should be a platform for our localization method.

Deliverable:

A drivable chassis.  

Resources:

Any and all needed components can be sourced from our locker in Troxler.  See Sean/Ahmed/Chris/Vijay for help.

SeanKetring, 2015-09-30:
I will add the battery holder and stack to the chassis

AhmedSamara, 2015-09-30:
Things we'll need:

- [Set up a BeagleBone](https://github.com/IEEERobotics/bot/wiki/BBB-Setup) (ask me for help on this if you have problems).   
   - Note: We need to figure out a better way to do this long term.    

- Wiring/battery's (ask @SeanKetring for help).

AhmedSamara, 2015-09-30:
(NEW PEOPLE)

please get in on this, we need help on this and need new people in on this.

jasteve4, 2015-09-30:
I plan to work on the bot wiring.

kvijay1995, 2015-10-01:
I will work on setting up the beaglebone and connecting to it via the android app

SeanKetring, 2015-10-02:
Still to do on this
<pre>
Finishing Chassis Tasks
  1)Finishing Motor Connections
       1.a)Tidy up wires ~ use electrical tape to secure free wires/connect motors to cape 
       1.b)Finish shaft couplers (outsourced)
       1.c) Attach mechanin wheels 
  
  2) Power Connection 
       2.a)Mount purple dark power board to underside of bot ~ if clearance permits 
       2.b)Finish Fuse and switch board (grey 3d printed part with wires coming off it)  
           2.b.i)Locating 2 suitable two terminal switches~Might have to buy if not in troxler
           2.b.ii)Locate 1 suitable 10 A fuse holder 
           2.b.iii)Mounting selected components into fuse and switch board
           2.b.iv)Soldering wire connections   ~  Should document once done  
           2.b.v)Attach board to open beam 

   3) Connecting the user 
          3.a)Mounting the usb hub
              3.a.i)Beagle bone usb connection ~ use a cable from the cable box
              3.a.ii)Dark power connection ~ 5v rail 
    
   4) Design review and testing 
           4.a) Test power up 
           4.b) Test motor connections through CLI 
           4.c) Test app based control 

Task complete 
</pre>

Deadline:   Next Sunday at Midnight 

<b>**New people**</b>  
This task is great for learning the basic structure of a bot, anyone and everyone is invited to contribute.  However some tasks will require a senior member to advise, for safety to equipment and ourselves!  Comment on this issue to state any intentions to work on this!

nv4, 2015-10-03:
I'll try getting done most of the stuff under Power Connection; soldering, mounting, etc.

Sidd-GrizzAp3, 2015-10-06:
I shall help out with this today.

SeanKetring, 2015-10-12:
Update:

Last things that need to happen for this

1) We need to tap the set screw holes for our new shaft couplers, at which point we will be able to mount our wheels, using our fancy new hardware courtesy of Touchstone 3D. 
 
2) We need to test the control of the bot through the CLI.

cdyoder, 2015-10-14:
I'm looking into tapping the couplers either Wednesday 10/14 or Thursday 10/15.

kvijay1995, 2015-10-16:
The server is up and running on the bot as of yesterday and the CLI is ready to be tested out with the motors (with or without the wheels). Maybe we'll have a bot drivable bot next Tuesday?

cdyoder, 2015-10-16:
Couplers are tapped and ready to go with the motors (require an M3x0.5 screw). Need to speak with @AhmedSamara @SeanKetring @PaladinEng @kvijay1995 @jasteve4 Ismail and Mark regarding attaching the couplers to the wheels Tuesday.