IEEERobotics/bot #395: Stereoscopic vision

SeanKetring, 2015-09-16:
Task:

Dump any and all information found from a lit review related to stereoscopic vision.   Questions to consider, how easy is it to implement, how expensive is the hardware, where can we get it, what are the power/data demands of the hardware.  

Deliverable:  

Information that can help lead to deciding whether or not this is a viable solution.  Posted on this issue, and reported at the next team all hands meeting.

AhmedSamara, 2015-09-16:
(Issue updated, please view it on github)

AhmedSamara, 2015-09-16:
@nv4

dannykhalil, 2015-09-17:
Going to be joining in on this

fliesnow, 2015-09-20:
After some cursory investigation, we found that opencv actually has a stereoscopic depth mapper included in its software.
https://opencv-python-tutroals.readthedocs.org/en/latest/py_tutorials/py_calib3d/py_depthmap/py_depthmap.html#py-depthmap

imaniaminah, 2015-09-22:
http://www.intorobotics.com/fundamental-guide-for-stereo-vision-cameras-in-robotics-tutorials-and-resources/