IEEERobotics/bot #396: Motor cape encoder reader

SeanKetring, 2015-09-16:
Task:

We need someone to determine if our current motor cape is capable of reading encoders, and if so, how to make use of the data.  

Deliverable:

A yes or no answer to the first question, and, if yes, steps to implement the device functionality.   Post to this issue and report at next team meeting.

AhmedSamara, 2015-09-16:
Issue updated, please view on github for most recent version.

tabarrett, 2015-09-18:
Update: I found some resources that might be helpful but I am still trying to figure out the details:
http://www.ni.com/tutorial/7109/en/ This is a concise how they work guide with some short examples.
Theres an example of people setting up one I found: http://letsmakerobots.com/content/how-use-quadrature-encoder. These might be helpful for people who understand the python code found in dmcc_motor.py

shreyashgotee, 2015-09-22:
I'm going to work on this one...

wtshaw, 2015-09-22:
I'm going to work on this.

imohame, 2015-10-21:
I'm working on this. So let's on Thurs. to discuss it

imohame, 2015-11-13:
This is the data sheet for the motors
https://www.pololu.com/product/2271

imohame, 2015-11-20:
[image]
Sean helped me figuring out the correct to match the pins of the cape and the micro-controller as shown in the figure. Now the encoders give values for the pos. and the vel. We need to see how to interpret them.

AhmedSamara, 2016-01-09:
We have an example of this being used from a few years ago:

https://github.com/IEEERobotics/bot/blob/master/bot/hardware/dmcc_motor.py

So calling the position, velocity, etc, properties is really straightforward.

I added a rapper in omni_driver.py on the omni dev branch.

AhmedSamara, 2016-01-21:
@WillDuncan1

AhmedSamara, 2016-02-01:
https://github.com/IEEERobotics/DMCC_Library/blob/master/examples/dmcc_motor_test.py

Here's a sample script of them being used