IEEERobotics/bot #397: Use Gyroscope/accelerometer/etc to correct robots motion.

AhmedSamara, 2015-09-18:
### Introduction   
Whenever the robot moves, especially when using Mecanum wheels, the robot drifts and moves unexpectedly. Some of our members have mentioned that in the past, they used a gyroscope to help correct this motion. Some members have used an IMU before, whichcontains a gyroscope, an accelerometer, and a magnetometer (aka compass).   
We don't currently own an IMU but are willing to invest in one if it proves worthwhile.   
                
### Notes     
* We have individual sensors as one unit if using the IMU, or multiple sensors, as a whole proves too difficult.  
* [Start by creating hardware abstractions for the sensors](https://github.com/IEEERobotics/bot/tree/master/bot/hardware)   
* Multiple people can start getting different sensors to read in raw values.   
   
   
### Requirements     
* Any amount of additionally stabilized movement would be helpful.   
* A fully robust system might make use of all 3 sensors, but using just one is acceptable if that proves to be too 

### Resources     
* There is a box of sensors in the locker. See if any of those are useful, or if we need to invest in new ones.

dmpage318, 2015-09-22:
I will work on this.

jonathonwpowell, 2015-10-02:
Working on getting data from the magnetometer in issue #410

tabarrett, 2016-01-13:
We tested the Magnetometer is was initially successful but we should make sure it works from the beagle-bone.  I will look into testing the Gyroscope.

tabarrett, 2016-01-20:
We had no Gyroscope but we did have just and 3 direction acceleratometer.  We tested it with the Ardurino and found we could get good measurements out of it but when we used integration to try and get distance we found there was so much error none the data was useful. We in our research for this were found people telling us this and we confirmed it was true. Here is the best resource we found on bot localization with IMU and their related devices: http://stackoverflow.com/questions/17572769/calculating-distances-using-accelerometer