IEEERobotics/bot #399: Create omni-wheel driver.

AhmedSamara, 2015-09-18:
### Introduction   

2016 competition involves very precise manipulations, which may mean that something unreliable and unstable like mecanum wheels might be a bad move. Omni wheels solve a similair function.  #161 is the task for physically building it.
It's possible to use them the same way as Mecanum wheels using [Mecdriver](https://github.com/IEEERobotics/bot/blob/master/bot/driver/mec_driver.py), as experimented with in #179 , but ideally we'd use them the 'right' way, where it turned and drove when needed.

### Requirements   
* a hardware abstraction and interface where it can be issued commands such as rotate, strafe, etc. Deciding the appropriate interface is a part of the task, but one with the same level of abstraction as Mec_driver is expected.   

### Resources    
* [Use mecdriver as a model](https://github.com/IEEERobotics/bot/blob/master/bot/driver/mec_driver.py)
   * Mecdriver uses the same hardware (motor, cape, and signal wise), just a different frame and wheels.      
* Talk to @ahmedsamara for help.

AhmedSamara, 2015-10-02:
@khweerak

dkaramitrov, 2015-10-06:
I am interested!

JacobDremann, 2015-10-06:
Interested

khweerak, 2015-10-06:
I am interested

AhmedSamara, 2015-10-20:
@EthanAnd 
@HaydenLawrence 
@Djvacto

imohame, 2015-10-21:
I'm going to set up the BB for this.

LaurenBrennan, 2015-10-27:
I'll work on this too

LaurenBrennan, 2015-11-03:
I am still interested but I wont be at the meeting tonight, could you guys update me when you finish tonight? Thanks!

AhmedSamara, 2016-03-09:
@kvijay1995 Did this and it's working pretty well.

closed 2016-03-09 by AhmedSamara