IEEERobotics/bot #400: Localization : Hacking and Modding a Mouse

Sidd-GrizzAp3, 2015-09-23:
I am exploring the utilization of a mouse's optical sensor and gathering information on how to hack / mod the sensor to interface with a serialized i/o protocol.  Please post discussions (if interested) and or research material on this topic.  

**Note: this likey cannot be a stand alone system, it has to be integrated at the system level with other proximity sensors or the like in order for more finetuned movement, and error mitigation.  **

Sidd-GrizzAp3, 2015-09-23:
Pros: 
mice are easily avialabe (no long shipping costs \ timing schedules), 
there are hardware specs, 
are cheap, 
abundant, 
has tons of information in relation to localization robotics. 
can be very very precise, if purchasing a good sensor. 

Draw backs: 
requires very tight physical constraints for good feedback,

AhmedSamara, 2015-09-23:
Hey Sidd, we already have an issue for this.

#350

closed 2015-09-23 by AhmedSamara