IEEERobotics/bot #402: Gripper Redesign

SeanKetring, 2015-09-25:
After testing the gripper that comes with the DAGU arm, it was apparent that it will not work for block grabbing and we are going to have to redesign the gripper to make it more appropriate to our competition.    

I'd like to know if anyone has any ideas on simple grabbing mechanisms that we could start prototyping?

PaladinEng, 2015-09-26:
I'm assuming that the problem was that the blocks were "tippy" and sliding
out of the pincers? Can you confirm?

My first suggestion is to consider expanding/replacing the pincer pieces
with longer wider pincers to increase the contact area.

SeanKetring, 2015-09-26:
Yes, the blocks would rotate around where the pincers were making contact, and then fall.

PaladinEng, 2015-09-26:
Also, you may want to line the inside of the new pincer pieces with a
 stiff, thin, & soft material, like one-sided foam tape, to increase the
likely-hood of even contact pressure.

Do you guys know how to design parts in solidworks or inventor?

SeanKetring, 2015-12-02:
Redesigned and working well!

closed 2015-12-02 by SeanKetring