IEEERobotics/bot #403: Issues with time out with  ssh private auth while using vagrant with virtual box.

Sidd-GrizzAp3, 2015-09-27:
If you're having issues with vagrant and virtual box where the ssh private auth is timing out and failing.  I think there is a pretty easy fix: you have to enable intel viritualization technology in bios for your particular cpu or the like if you're using amd, or the virtual box will not be function and up properly.

dfarrell07, 2015-09-28:
> ssh private auth is timing out and failing

Debugging-wise, this is very common and typically means the box has generally failed to boot. I think that's what you're seeing, as not having the virt extensions would prevent VMs from running in general.

Sidd-GrizzAp3, 2015-09-30:
Yea!  Problem is on my home desktop, I'm running 4.7 Ghz.   Which is fine and stable, until I turn on virtualization, go to sleep mode, wake and try to use any cpuID functions i.e. thermal sensors etc.   I get a driver error and bsod.   -___-
Sticking to vagrant on mac, wayyyy better.