IEEERobotics/bot #405: Pan-handling for money.

AhmedSamara, 2015-09-30:
Plexus has agreed to consider sponsoring us with mentorship, lab space, and funds.  

We just need to write them a letter.

So, open-source faeries, please drop in a well-written begging for money letter:

https://docs.google.com/document/d/19jB-Geqrne3_aSgaOHgC07-i3n0NOxm4tXMzJMSBcgs/edit

cdyoder, 2015-10-15:
A draft has been written, yet it still needs editing and proofreading. I would also be inclined to get the feedback of other senior members who may be skilled at writing professional letters asking for project funding. Below are some useful links:

http://www.plexus.com/
http://www.plexus.com/about-us/culture
http://www.plexus.com/community

dfarrell07, 2015-10-15:
> senior members who may be skilled

Gave it a quick read, made some small fixes.

Love the open source doc writing.