IEEERobotics/bot #406: Can we power the Kinect on Dark Power?

tabarrett, 2015-09-30:
So it looks like the Kinect research group is thinking that the Kinect could work as part of our system of sensors that handle localization. However, It was pointed out tonight that if we want to use it are we sure we can power it via Dark Power. The Kinect uses 12V and up to 1.08A. You can read that off the power pack that comes with it. I did not get to spend alot of time with it so someone correct me if there is more going with the Kinect Power requirements. There are tutorials on how to sever the current power cord and build an interface: http://wiki.ros.org/kinect/Tutorials/Adding%20a%20Kinect%20to%20an%20iRobot%20Create#DB25_Connector_on_the_Create
However Sean told me we don't have 12 Volt rail on Dark Power, and Josh mentioned in Passing that there won't be. So we might have a serious problem with powering the the Kinect. I don't know enough about Dark Power so people please chime in on how we can deal with this.

jasteve4, 2015-09-30:
Please see the new issue on [12 volt supply suggestions](https://github.com/IEEERobotics/bot/issues/407)

SeanKetring, 2015-12-02:
No longer being pursued as a feasible solution

closed 2015-12-02 by SeanKetring