IEEERobotics/bot #407: Need to develop the hardware to support a 12V rail.

jasteve4, 2015-09-30:
Dark Power has the ability to produce around 1.2 - 9 V due to the input voltage. The chip that we are using for dark power has a duty cycle limitation that prevents 100% duty cycle output.   

The input voltage comes from the battery, 12 - 16V.

To get 12 volts as an output, we will need to use a buck boost or a fly-back converter. 

I suggest we take a look at this on hack day.

For those of you that want to lean about power system this is a good start.  

Just a quit look at liner technology, I found the following dev board for a [buck-boost converter](http://www.linear.com/product/LT3958#demoboards)

Sidd-GrizzAp3, 2015-09-30:
Buck-boost would work even if the voltage input was coming in was lower than 12 V.  

I further did a constraint search for boost converters on linear technologies seen here: 
http://www.linear.com/parametric/Step-Up_%28Boost%29_Regulators#!1646_Buck-Boost|SEPIC!1032_%3C=12!1033_%3E=16!1105_%3C=12!1034_%3E=12!1035_3.32:16.6!1107_1:5!2086_%3E28!vinmin_12!vinmax_16!vout_12!iout_1!vout2_!iout2_!vout3_!iout3_

is this rail requirement only for Kinect?

jasteve4, 2015-09-30:
Some high end camera system need 12 Volts, so it may be a good idea to address this issue. We can always go with a off the shelf solutions.

SeanKetring, 2015-09-30:
Found some 12 VDC switching regulators on Google for <$20

This one has a 4 amp limit, which would probably be enough for a Kinnect and Camera, its 15 bucks, and could be here by the tenth 

http://www.batteryspace.com/dc-dcregulatormodule13-17vdcto12vdc4ampratewithremovableheatsink.aspx

This one bellow is less than 6 bucks but has a 600 mA max so not sure how useful it is

http://www.getfpv.com/12v-step-down-voltage-regulator.html?utm_source=google_shopping&gclid=CIeWx-Otn8gCFQ4WHwode6wJCA

This is a potential idea if we want to avoid making our own which would probably cost more and take longer, but would probably be more fun :)

Sidd-GrizzAp3, 2015-09-30:
LOL @jasteve4  and @SeanKetring's posts of not having fun.

jasteve4, 2015-09-30:
We currently use a 4 cell battery. The output voltage is 14.8 volts. If we use the 80% rule for safe use of Lipo batteries, than our batteries can discharge down to 11.84 V. So we have to use a buck boost or a flyback converter. Standard buck or LDOs converter have an operating range. Here is a good paper to read up on [limitation of a DC/DC buck converter](http://www.ti.com/lit/an/slyt293/slyt293.pdf)

Dark Power uses buck converters to produce voltage form vref to 87% of the input voltage. So, at best dark power can maintain 12 volts at 13.56 volts. So, that is around 92% of the max capacity of the battery. This means that shorter run times.

SeanKetring, 2015-10-01:
We never knowingly use our batteries at that level, and I personally have never seen our batteries get that low, and if we follow our <b>never leave batteries plugged in</b> rule, then we can plan on never seeing a voltage that low from the batteries.

jasteve4, 2015-10-07:
I Plan to test the Dark Power at 12 V on Thursday. I Think all I need to do is swap out the R2 resistor with a 1.2K ohms resistor and make sure the input voltages is above 14 V.

Sidd-GrizzAp3, 2015-10-07:
I can also have two test cases for the adc channel I'm trying to
configure.

closed 2015-12-02 by SeanKetring