IEEERobotics/bot #410: Magnetomenter for locallization

jonathonwpowell, 2015-10-01:
I am looking into the possibility of using a magnetometer (compass) as part of our localization strategy.   I believe it will be a helpful piece of the puzzle in figuring out where we are on the course.  Please respond with thoughts and opinions.

Information found so far:
https://www.sparkfun.com/products/10530

jonathonwpowell, 2015-10-01:
Ideas for use of the compass:
* Have the bot on start up calibrate to start position (as the starting position for the course is constant relative to the rest of the course
* Use the compass to confirm that the bot moves in a strait line/correct for accidental turning
* Use compass to calibrate where bot thinks it is a predetermined points (if the bot knows the angle it is at and what the distance should be to a wall at two angles from LIDAR it should be able to move itself to where it believed to be)

jonathonwpowell, 2015-10-02:
A magnetometer has been used tin a prior competition and had issues on-site for competition but not while testing at ncsu.  Need to make sure bot does not rely on compass to function properly

Sidd-GrizzAp3, 2015-10-02:
I was thinking of developing some sort of hybrid localization system: one for long-range: deadreckoning localization, and then mitigate error accruing: correct via the chosen computer vision (kinect or bumblebee or w.e.).   

I.e. always go to a known corner of the map as a "Pylon", can use proximity sensors and computer vision to verify at pylon.  Then we can use dead reckoning to navigate to the desired location approximately, and then once in the vicinity, can use computer vision with qr codes (which actually gives a lot of distance, and angle of viewing data) to "dock" to the desired exact position for desired action i.e. getting blocks, or dumping blocks.    I think dead reckoning would be easy to implement and not that cpu intensive and fast control of movements for big moves around the course.  For the more exacting movements we can use less error prone technologies.   

Having said this, I think this would be a good idea to use a compass as part of the dead reckoning!
@kvijay1995  @BrettGoldbach  @kennychuang  @AhmedSamara

tabarrett, 2016-01-13:
Sean, VJ and I all messed around with alternate localization devices we tested Sonics using arduinos using code similar to this: http://www.theengineeringprojects.com/2015/02/interfacing-multiple-ultrasonic-sensor-arduino.html.
We also tried using a magnetometer(compass): http://www.funnyrobotics.com/2011/03/arduino-with-hmc6352-digital-compass.html. Both worked but the sonics had limited range and were off by a centimeter. The magnetometer recording were similar to the value from our smartphone compasses