IEEERobotics/bot #414: Update PyBBB to handle pin exports and BBB setup.

AhmedSamara, 2015-10-21:
### Introduction   

Currently when someone wants to create a GPIO or PWM object, they use the [pyBBB library](https://github.com/IEEERobotics/pybbb) to create a GPIO object and talk to the hardware through that.

Before you can do that however, you need to export the pin by editing the [setupall.sh script](https://github.com/IEEERobotics/bot/blob/master/bot/setup/fs-common/root/startup/setup_all.sh), and then running the [setupbone.sh](https://github.com/IEEERobotics/bot/blob/master/bot/setup/setup_bone.sh) script, which copies `setup.sh` into init.d (amongst other actions) and makes it so that the pins get exported every time the beaglebone turns on.

I think it would be a lot more user friendly if all of this was handled within the object itself, unless @dfarrell07 believes there is a good reason for keeping it the way that it was originally done.

### Notes   
* There is another step that has to happen first: The device tree overlay which we currently handle manually. There are other libraries, such as the adafruit BeagleBone black library that may handle all of this itself and it may be worth it to just convert our codebase to use that instead.     

### Requirements    
* updated and improved pyBBB library that only requires one step to create new GPIO. 

 
### Resources   
* [Guide for exporting pins manually](http://kilobaser.com/blog/2014-07-15-beaglebone-black-gpios), may offer insight on what the object needs to do.         
* [BBB pins wiki](https://github.com/IEEERobotics/bot/wiki/BBB-Pins), on how we currently do things.

deepakiam, 2015-10-21:
I will do this

dfarrell07, 2015-10-21:
> I think it would be a lot more user friendly if all of this was handled within the object itself, unless @dfarrell07 believes there is a good reason for keeping it the way that it was originally done.

Off-hand, that seems to be a good idea that makes the design less complex by abstracting that concern into a single thing. I don't see any obvious problems you'll hit.

:+1:

deepakiam, 2015-10-28:
I have done most of the conversion from shell to python. Stuck in some places. I am new to python and there is loads of coursework currently and hence the delay. Sorry about that. Will try to make it this thursday.

AhmedSamara, 2016-01-08:
@relnor

AhmedSamara, 2016-01-08:
https://learn.adafruit.com/setting-up-io-python-library-on-beaglebone-black/installation-on-ubuntu

an alternative is to just find a different library to use.

relnor, 2016-01-08:
I checked out a beaglebone on thursday 1/8/2016, planning to bring it back by tuesday. thanks.