IEEERobotics/bot #417: Arm control abstraction and Kinematics

BrettGoldbach, 2015-11-04:
Senior design group needs to create an arm control abstraction using the Kinematics code linked below and likely consisting of 6 servo objects.
@AhmedSamara @BrettGoldbach
https://gist.github.com/BrettGoldbach/dc765967d92ae1810025

AhmedSamara, 2015-11-04:
So the issue here that we needed to discuss is how we're going to integrate the inverse/forward kinematics into our main codebase.

There are a few ways that we can structure this: 

My idea was that we'd split it into two parts: A `hardware` arm abstraction that consumes 6 servo objects (and is basically just a rapper for the sake of keeping code explicit), and a kinematic_solver object that handles position <-> join position calculations.

Another option is to just have one abstraction that does both, and you could argue that two abstractions is unnecessary because the object above does literally nothing but take the angle it's passed and forward it to the servo object.     
I mainly cared about this just because I wanted to have a clear differentiation between 'hardware communication' and 'logic', but as I types this up I figured the servo object already handles that.

WillDuncan1, 2015-11-25:
We tested to see if dark power could power the servo cape and it seems to be working just fine.  Seems to be drawing more than enough amps to effectively power all the servos.  Much better than using the lab power supply.