IEEERobotics/bot #418: Detecting the Start Signal

SeanKetring, 2015-11-05:
<b>Introduction</b> 

Our robot can't start if it can't detect the start signal.  We need a rig of the start signal, basically a couple LEDs mounted into the back wall of the competition course.  Then we need to use that rig to test our ideas for detecting the signal.  This task will be very multidisciplinary, and is good for a group of new folks.  We need someone to take this on.  

<b>Notes</b>
This task is extremely important, and it is a real responsibility to the competition bot.  

-Rig is needed to place LEDs in appropriate location relative to the start location 
-We will need to research options; some potential ideas, using the webcam on the arm, using some type of photoresistor, 
-We will need some way to integrate the detection system with the robot, electrically/mechanically/and in code. 

<b>Resources</b>

Competition rules are here 
https://docs.google.com/document/d/1ITIsL9fpTk5HKEJW1NENVrkgfCgXqONWpm0sevzeYmo/edit#heading=h.obp73v47rneu

SeanKetring, 2015-11-05:
Upon review of the rules this is what they had to say

Q: I am not sure how the robot is supposed to start. The rules have two different scenarios; the timer starts when 4 LEDs are turned off AND "Robots will simply start by pressing a "Go" button on the robot itself." Which way will the competition start?
A: The starting LEDs are removed from the rules. Instead of LEDs, robots will be activated during the start of each competition by using a rocker switch to be located on the robot. The switch must be visible to the judge but will be activated by a student team member at the direction of the judge at the start of the competition. Any other devices except for this physical rocker switch  used for starting the robot ( like a remote control) are prohibited. Once the rocker switch has been flipped, the robot must be completely autonomous.  The team should design for  two options for starting, depending on the assignment  to the left or the right side of the field.

AhmedSamara, 2015-11-09:
ie: There is no start signal now, we just have to make the robot start on powerup, and also read a GPIO for programming in east/west courses.

kvijay1995, 2015-12-23:
We've decided to use a 3 port slide switch to detect whether we need to do the east/west courses. The idea is to do the detection automatically when the robot is powered on.The schematic for this is as follows:

SeanKetring, 2015-12-23:
The switch has been added to the Motherboard PCB, this is the switch I am planning to use  

http://www.newark.com/eao/09-10290-01/switch-slide-spdt-co-500ma-though/dp/52F519

SeanKetring, 2016-01-26:
Start switch has been hardwired and is ready to be tested.  This task is very high priority, and fairly low hanging fruit, 

@kvijay1995  or @mynameis7  can one of you test this out at tomorrows meeting(1/26)?