IEEERobotics/bot #419: Investigate ROS for BeagleBone and/or

AhmedSamara, 2015-11-14:
## Introduction      

Currently, our code-base that has largely been built from scratch over the years by previous members of the team.    
Even though this was a fairly good/robust implementation for the time that it was written, it's starting to show it's age and requires a lot of maintenance, and using a non-standard solution presents it's own difficulties when people run into problems and have a hard time finding resources because there just aren't any people  besides us that use it.    

I don't know much about ROS, but given that it's a standard solution for solving the common problems of all robotics projects, which we do ourselves a lot, I think it's one that's worth exploring more.     

## Resources    
[ROS website](http://www.ros.org/about-ros/)     
[ROS implementation of BeagleBone Black](http://wiki.ros.org/BeagleBone)

AhmedSamara, 2015-12-22:
http://sdtimes.com/inside-the-robot-operating-system-the-robotics-industry-and-the-open-source-robotics-foundation/

Going to bump the priority of this because this actually looks pretty important.

AhmedSamara, 2015-12-30:
@kvijay1995

AhmedSamara, 2016-06-11:
@razcandy

BlackMamba591, 2016-06-14:
Ros is a highly user friendly framework. It has lot of libraries for beaglebone. Here is the wiki for that http://wiki.ros.org/BeagleBone