IEEERobotics/bot #424: Flaky I2C connectivity from BeagleBone.

AhmedSamara, 2015-12-03:
The BeagleBone to servo cape connection has been extremely flaky.   

The cape uses a PIC controller programmed to act as an i2c slave, and is [found here.](https://github.com/IEEERobotics/BBB-ServoCape)

We were successfully communicating between the BeagleBone and the servo-cape,but it periodically seems to go down for no clear reason, only to come back up for no apparent reason.  

All connections and headers seem to be secure.

Another strange symptom is that even though the address of the servo cape isn't showing up (0x10), the first time we run `i2cdetect -y -r 1`, we see a mysterious 0x03 device address, but never see it again afterwards.

There are no other devices plugged in, and we've also tried changing the address of the PIC. 

This address shows up once, but never again.
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- UU UU UU UU -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --      
```

AhmedSamara, 2015-12-03:
@PaladinEng @jasteve4 do you know what could cause this?

PaladinEng, 2015-12-03:
What does the I2C signal look like on the scope?

closed 2016-01-03 by SeanKetring

AhmedSamara, 2016-01-03:
Can we really say that this is closed?
We haven't run into it in a while but do we know what caused it or have a
way to prevent it in the future?

SeanKetring, 2016-01-04:
I really believe it was the connection through multiple headers that were not plugged in all the way, thats just my theory, we'll have to see if this issue comes back