IEEERobotics/bot #425: Design competition robot.

AhmedSamara, 2015-12-10:
Hey everyone, since it's nearly the end of the fall semester, I think it's time that we started discussing how we're going to assemble the final robot. We should at least have a design nailed down by the end of winter break, and we'll be having a meeting soon to discuss it, so please be prepared with ideas (and post them here).   
So far we have a few guidelines for how it needs to be assembled.

Keep in mind the requirements for the competition:    
*  cannot exceed 12” x 12” x 12” at the start of the match,
*  can expand to a maximum of 20” x 20” x 20” during the game.    
* Omni-wheels (unless someone has a good reason for not using them)    
* Must be able to mount Dagu arm       

@cdyoder and @jdekler  are currently working on a stable-chassis, would you guys mind discussing that as well?

SeanKetring, 2015-12-10:
I have talked with alot of folks and I have an idea of what all of our ideas together includes I have some sketches of the final design of the bot:

[image]
[image]
[image]

Picture quality is not great; sorry bout that.   We have to nail down what we are going to pursue as a final design this upcoming Tuesday, and then we need to build it over winter break.  We will be behind if we don't.

closed 2016-10-02 by AhmedSamara

adgooden, 2016-10-04:
Ahmed,

Can you remove me from the IEEE communications? Thanks

Andrew Gooden
North Carolina State University
Electrical and Computer Engineering
adgooden@ncsu.edu