IEEERobotics/bot #427: Lower Chassis Review

jdekler, 2015-12-18:
Here are the current lower chassis designs. If everyone could review and suggest changes that need to be made that would be great. 

[image]
[image]
[image]
[image]

PaladinEng, 2015-12-18:
what are the sets of three large holes in the corners of the acrylic for?

jdekler, 2015-12-18:
Wiring Holes. Mostly just for easier management. Now that Chris has posted the upper and lower bot it is apparent that the holes will move. I tried to place them close to where the vertical support beams would be so that it would be easy to run wire up and down.

SeanKetring, 2015-12-23:
Acrylic was laser cut today, we are ready to start mounting motors. Do you have all the hardware you need to do this, or do we need to order some stuff @jdekler ?

AhmedSamara, 2015-12-30:
https://drive.google.com/drive/u/0/folders/0B-T8lr4gzD9lU2NzeHYzbklDa3M

So we ran into some compatibility issues between solidworks and the software that my parents use for their laser cutter, so we had to do some conversions. Could anyone check to see if this looks right?   

Also @cdyoder @jdekler how important is it that all of the cut is aligned in the exact locations, assuming they're all still the same relative to each other?  

We can't align the edges like they do on any fancy laser cutters, so we're not going to use the acrylic sheets you gave me and just use bigger ones and cut a 12x12 square out of them. Is it too late to return the sheets we have?

cdyoder, 2015-12-30:
The important piece is that all holes are alligned relative to each other.
If so, we can make it work, although it may not be easy/pretty.

Let's save the acrylic as I forsee us having other uses for it.

AhmedSamara, 2015-12-30:
Is it normal for acrylic to catch on fire and melt?

cdyoder, 2015-12-30:
Umm......no?

How badly damaged is the piece? Got a picture?

Sincerely,
Christopher D. Yoder
UGSA MAE Representative
North Carolina State University

AhmedSamara, 2015-12-30:
Just kidding about the melting.
There was some fire and more melting than I expected, but only on the
removed bits, the board looks fine.
We do however have a small problem. At some point in the process of
converting between file types and putting them in a file together (so I
could cut them from the same sheet) one of them became smaller than the
other.
All of the holes are in the same place, but the edge and the mounting slant
thing got cut off.
Umm......no?

How badly damaged is the piece? Got a picture?

Sincerely,
Christopher D. Yoder
UGSA MAE Representative
North Carolina State University

> Is it normal for acrylic to catch on fire and melt?
>
> > The important piece is that all holes are alligned relative to each
> other.
> > If so, we can make it work, although it may not be easy/pretty.
> >
> > Let's save the acrylic as I forsee us having other uses for it.
> >
> > >
> https://drive.google.com/drive/u/0/folders/0B-T8lr4gzD9lU2NzeHYzbklDa3M
> > >
> > > So we ran into some compatibility issues between solidworks and the
> > > software that my parents use for their laser cutter, so we had to do
> some
> > > conversions. Could anyone check to see if this looks right?
> > >
> > > Also @cdyoder <https://github.com/cdyoder> @jdekler
> > > <https://github.com/jdekler> how important is it that all of the cut
> is
> > > aligned in the exact locations, assuming they're all still the same
> > > relative to each other?
> > >
> > > We can't align the edges like they do on any fancy laser cutters, so
> > we're
> > > not going to use the acrylic sheets you gave me and just use bigger
> ones
> > > and cut a 12x12 square out of them. Is it too late to return the
sheets

cdyoder, 2015-12-30:
How much smaller?

Sincerely,
Christopher D. Yoder
UGSA MAE Representative
North Carolina State University

AhmedSamara, 2015-12-30:
Is the picture not sending?

cdyoder, 2015-12-30:
I do not see one either in my Gmail inbox or here on Github.

AhmedSamara, 2015-12-30:
[image]

It's about an inch smaller, but at least one of them came out nicely.

I also went ahead and cut some vinyl outlines of this so we can manually drill holes on the sheets we already have if there are more issues, so we should be good.

SeanKetring, 2016-01-08:
Lower chassis has been built to spec, and unless anyone objects, we can commit to using it in the competition bot.

closed 2016-01-08 by SeanKetring