IEEERobotics/bot #428: Entire bot update

cdyoder, 2015-12-18:
As a result of today's work, the entire bot (minus some stuff) is now on Github! The model in it's entirety can be found within the CAD repo. There is a read me text file (ASSEM_READ_ME.txt) which outlines the assembly files to open depending on the portion of the bot you wish to look at (for example, there is an assembly with the entire bot, the upper half, the lower half, etc.). The read me is not complete as of now but I will update it as we go throughout the next two weeks. 

One will note that there is work to be done on the upper half of the bot as all of the parts are not designed yet. That progress will be made as we go through the next two weeks. If anyone is able to 3D print, there are three files on the Google Drive (IEEE Ground Robotics -> 2015-2016 -> TO PRINT) which may be printed. 

Finally, as it stands right now, WE FIT WITHIN THE 12" X 12" X 12" BOX!  :D just fyi.... Below are two images of the model as it is saved. A design review is imminent and expected before the end of winter break. 

[image]
[image]

PaladinEng, 2015-12-19:
How much clearance do we have between the robot and the "box" ?

cdyoder, 2015-12-19:
The middle sheet of acrylic is almost 12"x12", so not much. Maybe 1/32"-3/32" on each side?

PaladinEng, 2015-12-19:
Ok. How big is the tunnel?

cdyoder, 2015-12-19:
The tunnel is 18"x18" in cross section I believe. Someone should  double
check though.
-Chris

cdyoder, 2015-12-22:
Most recent version of the CAD model of the robot. The rough design of the upper half is complete and the parts are on order. Also, the dxf files are being generated by James as this issue is being posted. Pics are below:

[image]
[image]
[image]
[image]

SeanKetring, 2015-12-23:
Model looks great!  Good job

closed 2015-12-23 by SeanKetring