IEEERobotics/bot #429: Winter break meeting times.

AhmedSamara, 2015-12-23:
A lot of people are planning on working during the break.    
For the sake of preventing people from having to work alone, please check in here whenever you get to the lab, especially if you're someone that has lab access so that other people can know whether or not they'll have someone to work with when they come in.

AhmedSamara, 2015-12-23:
Wednesday 12/23:

Ahmed Samara in @ 10:00

Vijay, Sean, and Will should be coming in later too.

jasteve4, 2015-12-23:
Are you all done for the day? I am in the lab now.

Josh Stevens @ 2.22

SeanKetring, 2015-12-23:
I'm here with Josh.  Will be here till 7pm or so

SeanKetring, 2015-12-23:
Anyone else coming in today?

AhmedSamara, 2015-12-28:
Ahmed in @ 11:00

The doors next to the oval are locked (even to card access), but the entrance next to the bus stop is unlocked.

cdyoder, 2015-12-29:
Yoder in Troxler @ 0930.

I didn't try the doors by the oval but the doors nearest the bus stop are open. Also, if the deck is closed, does that mean free parking on campus within the deck?

PaladinEng, 2015-12-29:
I parked yesterday on Oval Drive and did not get a ticket

cdyoder, 2015-12-31:
Yoder, Samara, Ketring, Stevens in Troxler since ~1200 12/31/15

SeanKetring, 2016-01-03:
Plan is to meet tomorrow at 12 in Troxler!

cdyoder, 2016-01-05:
Yoder, 12:04pm, 1/5/16

closed 2016-01-05 by AhmedSamara