IEEERobotics/bot #431: Segfault when importing Zbar on BeagleBone Black.

AhmedSamara, 2015-12-31:
It doesn't get very far. I tried to install zbar by downloading the code and running `python setup.py install`, but it doesn't seem to be working.      
```
root@bot:~/zbar-code/python# python
Python 2.7.3 (default, Mar 14 2014, 17:55:54) 
[GCC 4.6.3] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import zbar
Segmentation fault
```

AhmedSamara, 2015-12-31:
Trying to install zbar manually with `python setup.py install` gave this error:    

`command 'gcc' failed with exit status 1`

AhmedSamara, 2015-12-31:
Similair issue for RaspBerry Pi, it looks like the issue is related to not being comnpatible with the processor.

AhmedSamara, 2015-12-31:
Here's the full error

```
root@bot:~/zbar-code/python# python setup.py install
running install
running build
running build_ext
building 'zbar' extension
gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c zbarmodule.c -o build/temp.linux-armv7l-2.7/zbarmodule.o
zbarmodule.c:55:25: error: ‘ZBAR_DATABAR’ undeclared here (not in a function)
zbarmodule.c:56:25: error: ‘ZBAR_DATABAR_EXP’ undeclared here (not in a function)
zbarmodule.c:58:25: error: ‘ZBAR_CODABAR’ undeclared here (not in a function)
zbarmodule.c:62:25: error: ‘ZBAR_CODE93’ undeclared here (not in a function)
zbarmodule.c:74:25: error: ‘ZBAR_CFG_UNCERTAINTY’ undeclared here (not in a function)
zbarmodule.c:74:5: error: initializer element is not constant
zbarmodule.c:74:5: error: (near initialization for ‘config_defs[6].intval’)
zbarmodule.c:82:25: error: ‘ZBAR_MOD_GS1’ undeclared here (not in a function)
zbarmodule.c:82:5: error: initializer element is not constant
zbarmodule.c:82:5: error: (near initialization for ‘modifier_defs[0].intval’)
zbarmodule.c:83:25: error: ‘ZBAR_MOD_AIM’ undeclared here (not in a function)
zbarmodule.c:83:5: error: initializer element is not constant
zbarmodule.c:83:5: error: (near initialization for ‘modifier_defs[1].intval’)
zbarmodule.c:88:25: error: ‘ZBAR_ORIENT_UNKNOWN’ undeclared here (not in a function)
zbarmodule.c:88:5: error: initializer element is not constant
zbarmodule.c:88:5: error: (near initialization for ‘orient_defs[0].intval’)
zbarmodule.c:89:25: error: ‘ZBAR_ORIENT_UP’ undeclared here (not in a function)
zbarmodule.c:89:5: error: initializer element is not constant
zbarmodule.c:89:5: error: (near initialization for ‘orient_defs[1].intval’)
zbarmodule.c:90:25: error: ‘ZBAR_ORIENT_RIGHT’ undeclared here (not in a function)
zbarmodule.c:90:5: error: initializer element is not constant
zbarmodule.c:90:5: error: (near initialization for ‘orient_defs[2].intval’)
zbarmodule.c:91:25: error: ‘ZBAR_ORIENT_DOWN’ undeclared here (not in a function)
zbarmodule.c:91:5: error: initializer element is not constant
zbarmodule.c:91:5: error: (near initialization for ‘orient_defs[3].intval’)
zbarmodule.c:92:25: error: ‘ZBAR_ORIENT_LEFT’ undeclared here (not in a function)
zbarmodule.c:92:5: error: initializer element is not constant
zbarmodule.c:92:5: error: (near initialization for ‘orient_defs[4].intval’)
error: command 'gcc' failed with exit status 1
```

AhmedSamara, 2015-12-31:
Fixed using this Method:

https://www.raspberrypi.org/forums/viewtopic.php?f=32&t=76674

```
$ wget https://github.com/npinchot/zbar/archive/master.zip
$ unzip master.zip 
$ cd zbar-master
$ sudo python setup.py install
```

Not sure why this didn't work from the original zbar library.

AhmedSamara, 2016-02-03:
Apparently this still doesn't work?     

#450 ran into problems where it can't install things due to memory issues. It might still be related to this.

@BrettGoldbach Please update when you figure out why it won't work.

BrettGoldbach, 2016-03-24:
Before you run the setup.py install in the above commands you must run a apt-get install libzbar-dev to get the required zbar.h file for setup.py.

closed 2016-03-24 by BrettGoldbach