IEEERobotics/bot #434: Wifi Modules for the bot

SeanKetring, 2016-01-05:
To get around the consistant issues with the wireless NICs.  We think it would be worth the time and effort to look into developing an alternate solution.  One idea is to follow the example of ECE 306 and to use an IOT wifi module.  

End goal of this issue would be a wifi module that could be mounted and hard wired into the motherboard.  

More info on this later.

tabarrett, 2016-01-14:
Are we considering using the exact same module? I could work on this some I guess either way.

PaladinEng, 2016-01-29:
For those of us who went through 306 pre-wifi module, would you mind linking to the module?

SeanKetring, 2016-01-29:
@tabarrett can you post the part number?

adgooden, 2016-01-29:
All,

Are you all looking for the wifi module that Mr Karlson used in his 306 class? 

Andrew Gooden
North Carolina State University, Undergraduate
Electrical and Computer Engineering
adgooden@ncsu.edu

SeanKetring, 2016-01-29:
Yeah, do you know the part number?

adgooden, 2016-01-29:
I don’t have it. I took 306 when Mr Karlson still used GPS. But Im sure he could provide the part number if I asked nicely.

Andrew Gooden
North Carolina State University, Undergraduate
Electrical and Computer Engineering
adgooden@ncsu.edu

kvijay1995, 2016-01-31:
The device model is the STMicro SPWF01SA Wifi module. Here is the link I used to buy it:

http://www.digikey.com/product-detail/en/SPWF01SA.11/497-14636-ND/4809481