IEEERobotics/bot #438: LIDAR Squirrels and Other Problems

adwiii, 2016-01-13:
After a lot of successful work with the LIDAR before break, we came back to find that it is no longer giving us location data. Last week we were able to send and receive commands with the LIDAR , though most of them failed. Notably, the "Wanderer" command, according to the [XV-11 Hacking wiki](https://xv11hacking.wikispaces.com/LIDAR+API+Commands) the command is supposed to result in an ASCII art squirrel followed by "SUCCEEDED":
    .-"""-. .-"""-.
   /,:;;:;    ,'=. \
  : ;:;:;' .=" ,'_\ ;
  : ':;:;,/  ,__:=  ;
   \ ';;:;  =./)_  /
    `.`"=\_  )_"`.'
      `.   ``'".'
        `.   .'
          `.'
 
SUCCEEDED

When we ran this command we got this lovely picture and message:
[image]
Which has the message at the bottom "SQUIRREL CONTAINMENT FAILED"

We have no real knowledge of what could have caused this. We opened up the LIDAR and found this:
[image]
The FET in the picture looks blown, which might be the issue.

SeanKetring, 2016-01-13:
Any chance you can pull a part number off of that?

AhmedSamara, 2016-01-13:
I couldn't tell in person, it looks like the part numbers are cleaned off, but I found a [BOM for the XV11](https://xv11hacking.wikispaces.com/XV-11+PCB+Information).    

However this might be a different version, because the U5 in this one is [A3950 motor driver](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&ved=0ahUKEwiJ78KaoqfKAhXEZCYKHYbYBNEQFggdMAA&url=http%3A%2F%2Fwww.allegromicro.com%2F~%2Fmedia%2FFiles%2FDatasheets%2FA3950-Datasheet.ashx&usg=AFQjCNGxiQuibkf8GYO91LO2bHwAhkO-iQ&sig2=G99_SD1fqij1Z8uwB3VFeg&cad=rja) which has a totally different footprint.

AhmedSamara, 2016-01-13:
Is there any way to figure out what revision of the PCB ours has? They have a few different versions on it. 

I can't be sure, but a similair part on the other PCB's that looks like it could fit here is [NC7SZ14](https://www.fairchildsemi.com/datasheets/NC/NC7SZ14.pdf).

adwiii, 2016-01-15:
Upon further investigation, the board Ahmed kinks above are for the main board of the neato itself, not the lidar. We found someone else who did a teardown of the same board as ours of the lidar. http://diydrones.com/m/blogpost?id=705844%3ABlogPost%3A1591235
From that we can see the revision of board we have. The broken part of our board is labeled PHUI on that board, we are looking into seeing if we can find that part online.

tvbarnette999, 2016-01-15:
We found the datasheets for two parts made by TI that have the same labeling on the outside of the chip and the same footprint. We could request a free sample from TI since they are normally sold in minimum quantities of 250. Idk if that is something we could look into.

http://www.ti.com/lit/ds/symlink/tps793.pdf
http://www.ti.com/lit/ds/symlink/tps730.pdf