IEEERobotics/bot #442: Using OpenCV for localization.

AhmedSamara, 2016-01-18:
After much struggling, it seems that we won't be able to use the modified [QRCodeStateEstimation](https://github.com/IEEERobotics/QRCodeStateEstimation) library on the final robot, due to the ZBAR library being incompatible with the BeagleBone, as well as our struggling to get BOOST library working to work.

Looking at the source code for that library though, it's not that complicated and we should be easily recreated in python.    
All you really need is a set of (known) points, and to be able to identify them in the images captured by the webcam. 

This is the part that made Zbar really simple. When it finds a QR code, it also tells you where the vertices are, and this is the part that I'm running into problems with now.

I was hoping to solve this problem using [SURF points](http://docs.opencv.org/3.0-beta/doc/py_tutorials/py_feature2d/py_surf_intro/py_surf_intro.html)

The change in how we're approaching this is explained in my senior design presentation:

[image]

Basically the problem I'm running into is that even though I can now identify a bunch of known points on the QR code, there's still a few problems:    

* Not all of the SURF points are unique to each QR code.    
* When we find SURF points on the cameras captured frames, there are points on everything.   
How do we isolate the points just on the QR code?    
* Is there something better than surf points to use?

AhmedSamara, 2016-01-22:
Found this:

http://docs.opencv.org/3.0-beta/doc/py_tutorials/py_feature2d/py_matcher/py_matcher.html

AhmedSamara, 2016-01-22:
This looks promising now

https://gist.github.com/AhmedSamara/0ce6a836dd45e16b68ba

AhmedSamara, 2016-01-24:
[image]

It looks pretty promising.

PaladinEng, 2016-01-25:
Is this running on the BBB?

AhmedSamara, 2016-02-10:
http://stackoverflow.com/questions/7263621/how-to-find-corners-on-a-image-using-opencv

Here's another strategy for generic objects.

AhmedSamara, 2016-02-24:
Two cameras might help, here are instructions for setting up the second:

http://ubuntuforums.org/showthread.php?t=168221