IEEERobotics/bot #443: Massive oversight: The generic blocks do not have QR codes on them, no way to calculate pose.

AhmedSamara, 2016-01-20:
[image]

The generic blocks do not have QR codes on them.

This is the method that we've been using for calculating the vector from the end-effector to the block, without that QR code we don't have a way to put the end effector in the correct place.    

So we really don't have a way to handle this. The rest of the blocks should be possible to handle with Sean and I's senior design project, but that's a much higher risk area.

I'd hate to miss out on all of the points from what really should be a low-risk area.

AhmedSamara, 2016-01-20:
@adwiii @tvbarnette999 @dmpage318 Can you guys try to work on a solution for this problem since lidar is currently stopped?

AhmedSamara, 2016-01-20:
Using surf points to identify them as I was planning on doing in #442 because the edges of the blocks are so rough that the surf points are non-unique every time, we'd have to be able to look at every single block.

Since I'm re-working the vision anyway, I'm going to see if I can just identify the outline of blue objects and work from there.

AhmedSamara, 2016-01-22:
https://gist.github.com/AhmedSamara/0ce6a836dd45e16b68ba

Can fuilford-bros upload the code they had too? Both of these look promising for arbitrary object tracking.

adwiii, 2016-02-05:
[https://gist.github.com/adwiii/a12c523b0d675d25c45d](url)

This is the code that @tvbarnette999  and I were working on. It takes whatever blocks it sees and attempts to coerce them into a rectangle and whether or not it can do that it displays the angle that it finds. The coercion is just to make the angle better.