IEEERobotics/bot #445: Programming subroutines for arm using 7th degree of freedom

SeanKetring, 2016-01-26:
Certain functionality is ready for coding, 

such as:
 
1. Emptying the block hopper 
2. Positioning the arm accurately along the pinon.
3. Slow and controlled movement of the servos

@WillDuncan1 I think this is something that you and I need to work on together

AhmedSamara, 2016-02-02:
Since DOF-7 is attached to a DC motor, we'll have to use the motor cape encoder feedback. Refer to #396 for help with that. Most important thing is the [pyDMCC example](https://github.com/IEEERobotics/DMCC_Library/blob/master/examples/dmcc_motor_test.py). It shows all of the functions that you need. The only thing that really changes there is the way they're declared. He does a weird autodetect thing where it looks for i2c objects and then creates objects out of those.

The data structure is of the form:   dmccs= [ [motor1, motor2]
                                                                     ,[motor1, motor2]]

In other words, it's a nested list. dmcc's is all of the capes it found, and each cape contains 2 motors. You can see how it accesses those lower in the file.  

Keep in mind that the encoders can accurately tell you the position, but you can't always set it with the same accuracy. We can use the [PID controller](https://github.com/IEEERobotics/DMCC_Library/blob/master/examples/dmcc_pid_test.py) to set it with some accuracy but users have had issues with this in the past so we might define our own.

You'll want to start by figuring out what motor positions correspond to positions on the rail (I recommend finding two points and then just doing a linear fit so we have something simple of the form y = m*x+b), and then we can go from there with setting motor positions.

AhmedSamara, 2016-02-02:
Also another note relevant to #396 

The examples use a weird autodetect function.  `dmccs = pyDMCC.autodetect()` 

Where they just create instances with whatever's plugged in.

You can declare a single instance of the DMCC with:

```
import pyDMCC
my_motor  = DMCC(address)
```

Now this will create a DMCC motor object. In this situation, you'll want to look at one motor at a time instead of iterating through it like the example does.

closed 2016-03-31 by SeanKetring