IEEERobotics/bot #447: Help with setting up ZMQ server-Client setup

zabedwan, 2016-01-31:
Hey guys, I'm trying to setup a ZMQ server-client setup for an iOS App that will be controlling the BB8 bot. Currently we're using ssh to call on scripts in the beaglebone to perform an action. However, we're experiencing significant lag from app button press to action while using an iPhone. Any tips/links on how to get this setup working?

AhmedSamara, 2016-02-10:
The [ctrl_server](https://github.com/IEEERobotics/bot/blob/master/bot/server/ctrl_server.py)  builds a ZMQ socket, buids all of the subsystems (each piece of hardware or functionality is represented by a class), and then it listens on the socket for JSON messages, which it then decodes as a python dicts.

That's how it calls commands, it expects something of the format:
`obj_name function arg1:value1 arg2:value:2`

On the other side of that, you have the [generic client](https://github.com/IEEERobotics/bot/blob/master/bot/client/ctrl_client.py) which actual clients like the [command line](https://github.com/IEEERobotics/bot/blob/master/bot/interface/cli.py) or the [pilot](https://github.com/IEEERobotics/bot/blob/master/bot/pilot.py) use.

I'm not 100% sure this is the right format for your project, but this is the general structure of it. You're probably going to want something with socket programming though, and it might be an option to look into bluetooth as well.

AhmedSamara, 2016-02-26:
'helped

closed 2016-02-26 by AhmedSamara