IEEERobotics/bot #450: BeagleBone Space error

BrettGoldbach, 2016-02-03:
cannot copy extracted data for './usr/share/doc/libice-dev/changelog.gz' to '/usr/share/doc/libice-dev/changelog.gz.dpkg-new': failed to write (No space left on device)
Selecting previously unselected package libsm-dev:armhf.
Unpacking libsm-dev:armhf (from .../libsm-dev_2%3a1.2.1-2_armhf.deb) ...
dpkg: error processing /var/cache/apt/archives/libsm-dev_2%3a1.2.1-2_armhf.deb (--unpack):
 cannot copy extracted data for './usr/share/doc/libsm-dev/changelog.Debian.gz' to '/usr/share/doc/libsm-dev/changelog.Debian.gz.dpkg-new': failed to write (No space left on device)
Selecting previously unselected package libpixman-1-dev.
Unpacking libpixman-1-dev (from .../libpixman-1-dev_0.26.0-4+deb7u1_armhf.deb) ...
dpkg: unrecoverable fatal error, aborting:
 unable to flush /var/lib/dpkg/updates/tmp.i after padding: No space left on device
E: Sub-process /usr/bin/dpkg returned an error code (2)

After checking memory left we had about 165 MB left.

BrettGoldbach, 2016-02-03:
We changed to a new beagle bone and started from scratch. Now we get the following error hen we run the qr reading program:

Traceback (most recent call last):
  File "qr_feed.py", line 48, in <module>
    pil_im = Image.open('buffer.png').convert('L')
  File "/usr/local/lib/python2.7/dist-packages/PIL/Image.py", line 2295, in open
    % (filename if filename else fp))
IOError: cannot identify image file 'buffer.png'

AhmedSamara, 2016-02-03:
I think it doesn't have drivers installed.
It connected automatically before so I'm not sure why it didn't work this
time when it had internet access.

closed 2016-10-02 by AhmedSamara