IEEERobotics/bot #454: I2C Issues with Servo Cape and other devices

SeanKetring, 2016-02-07:
Some bug in the I2C routine on the servo cape is causing the bus to fail.  If the user inputs a read command to the servo cape it will break the communication.    So to use anything in the stack, do not run I2C detect and do not attempt to read from the Servo Cape.    This is a way to get around this bug, until we can figure out what it is.

AhmedSamara, 2016-02-10:
When you run an `i2cdetect -y -r 1`, the read is extremely slow, and no devices are detected,

It looks like the issue is beign caused by the SDA being held low, but we're not sure why or how to fix it.

similair issues happening here  
http://lm-sensors.lm-sensors.narkive.com/FFQ2T3ZL/i2cdetect-running-slow-but-not-bus-inactive-slow-not-sure-why

AhmedSamara, 2016-02-10:
According to the [TI I2C troubleshooting guide](http://www.ti.com/lit/an/scaa106/scaa106.pdf):

>Adding more I 2C and SMBus devices on the bus may exceed the 400-pF limitation. I
2C multiplexors, I2Cswitches, and I2C buffers and repeaters can isolate slave devices that are not currently needed to reducethe overall system loading, and then meet the maximum load capacitance specification.

AhmedSamara, 2016-02-10:
Never-mind, it's not a hardware issue.

Removing the I2C module from the PIC on the servo cape fixed the problem. It's definitely the servo cape holding the line low for some reason. 

@jasteve4 just notified us that there's a bug in some PIC controllers with I2C buses that force themselves low, so there's that to look into.

Still no idea why that's going wrong.

Removing it and putting it back in fixed it.

SeanKetring, 2016-03-31:
PIC was found to have silicon defects, switched to a new processor

closed 2016-03-31 by SeanKetring