IEEERobotics/bot #458: Simplify BeagleBone setup and code deployment using Ansible.

AhmedSamara, 2016-02-10:
## Introduction   

Currently writing and runnning code on the BeagleBone involves one of two different methods:

- Writing code locally and using rsync to bring it over (good but occasionally tedious without aliases and scritps)   
- Writing code directly on Bone (bad)    
- Writing code locally, pushing it to github, and pulling it on the BBB (very bad!)    

## Requirements   
A reasonably robust way to set up the BeagleBone without getting internet access on it.

mynameis7, 2016-02-12:
could we push code to its ip via git?  doesnt gits decentralized nature work well with things like this?

AhmedSamara, 2016-06-11:
We are now using Raspberry pi, but it would be good to have a setup that works on both.

mynameis7, 2016-07-20:
If we setup a git server on the bot (maybe try gitlab community edition https://about.gitlab.com/) then we could add the bot as a remote, so then we can allow pushes directly to the bot when connected over USB or something like that.  I'm not sure how PI handles a usb connection, but we could still use some of the spare routers lying around to provide an IP connection I think.