IEEERobotics/bot #463: Wheels keep falling off.

AhmedSamara, 2016-03-02:
Even with the JB weld, the wheels just fell off the motor again.

It looks like the problem is just that it's increasingly difficult to get the epoxy in the correct place, especially without messing up the motors.

Any ideas?

@cdyoder @jdekler

PaladinEng, 2016-03-02:
We're already working on new couplers. In the meantime, the shaft of the
motor needs a small divot drilled into it that will allow the set screw to
actually sit in the divot.

SeanKetring, 2016-03-03:
I'm going today to look at a specialty store for set screws to meet the specs of the part Chris suggested.  

http://www.amazon.com/Bolt-Base-Stainless-screws-Socket/dp/B00SN3532Q/ref=sr_1_5?ie=UTF8&qid=1456925346&sr=8-5&keywords=cone+set+screw+M3

I'm going to try and find an M3 and one size up in case we need to redrill and tap the set screw holes.

PaladinEng, 2016-03-03:
What's the store?

SeanKetring, 2016-03-03:
The store was wilders nuts and bolts and unfortunately they dont have a cone point set screw.  I think I found a much cheaper alternative to the one you sent Alwyn, on McMaster, which can ship today if we order fast.  Take a look at this, and let me know if you see any reason I shouldn't order this, if I don't hear back I'll go ahead and order them by the end of the business day.   They come in packs of 25 so there should be enough for the senior design teams as well.    

http://www.mcmaster.com/#91217A055

@cdyoder  @PaladinEng

cdyoder, 2016-03-03:
Those should work.
-Chris

PaladinEng, 2016-03-03:
Agreed, those are suitable. With a 25 pack, there should be enough for all
the senior design projects and the competition bots