IEEERobotics/bot #465: Changed IR capes from upper bot to lower bot

kvijay1995, 2016-03-06:
Earlier today, when me and @relnor were testing the navigation of the lower bot, we ran into some trouble with the I2C bus. Something on the bot held the SCL line low thereby preventing us from communicating to the motors and the IR cape. Upon further investigation we found the problem to be the IR cape although we're not sure what caused it start malfunctioning.

For now we've removed this cape (the cape with the green wire soldered onto it) from the lower bot and transferred the cape from the upper bot to the lower bot, since the upper bot isn't using the IR cape for anything as of now.

The problem with the malfunctioning IR cape is still to be investigated. Josh seems to think its the PIC.

SeanKetring, 2016-03-07:
We have some extra 1718 PICs in the locker

SeanKetring, 2016-03-31:
Extra IR capes are made,

closed 2016-03-31 by SeanKetring