IEEERobotics/bot #466: Meeting Times Over Spring Break

SeanKetring, 2016-03-07:
Use this thread for coordinating meeting over spring break. 

Will be in the lab tomorrow by late morning ~ noon

To do:
-Replace IR-Cape PIC 
-Continue programming arm subroutines
-Reconfigure IR sensor mount

Could probably use a wiring diagram of the bot at some point, any new electrical devs want to handle that?  ~ @tabarrett 

@kvijay1995 @relnor  How's navigation going?  Did the wheels fall off?

kvijay1995, 2016-03-07:
The wheels have been excellent. They haven't fallen off yet. Navigation development came to a standstill on Saturday since the IR-Cape caused I2C problems. Hoping to continue development tomorrow.

mynameis7, 2016-03-08:
Will anyone be working later on in the break?

PaladinEng, 2016-03-08:
I'll be in tomorrow at 6:30. I need to hear from James about the new
couplers

SeanKetring, 2016-03-08:
I'll be in tomorrow as well, probably by 1pm or so

kvijay1995, 2016-03-08:
@mynameis7 We probably will be working most of this week.

I'll be in today at around 2.30pm

mynameis7, 2016-03-10:
Is anyone coming today? I will be on campus till about 4

SeanKetring, 2016-03-10:
Will be there around 1:30

SeanKetring, 2016-03-11:
Hackday is in full swing

SeanKetring, 2016-03-11:
I can come in today After 3 if anyone wants to work together

AhmedSamara, 2016-03-11:
Now that the robot has been merged into two parts, it's difficult for people to work on subsystems at the same time.

So we have a calendar for people to help schedule meeting times outside of the normal meetings.

We're only 2 weeks away from competition, and we're extremely close to being done. We need everyone to do what they can with this final push to make sure we can compete!

https://calendar.google.com/calendar/render?pli=1#main_7

closed 2016-03-14 by SeanKetring