IEEERobotics/bot #471: Things to bring to SEcon 2016.

AhmedSamara, 2016-03-31:
If you think of anything we should bring, please list it here.  

Don't trust yourself to remember on the day of, if you think of anything at all, write it down.    

- JB weld (super important).     
- Backup couplers @jdekler   
- Backup cameras

SeanKetring, 2016-03-31:
-The Robot
-The Course
-Battery and charger 
-Box of misc. tools 
-Backup PCBs (white box in locker) 
-Tackleboxes 
-PIC kit red debugger 
-Beagle bone power cable, mini usb cables 
-extra motor, bracket and coupler 
-Backup 3d printed parts
-Backup IR sensors
-Backup Omni wheels 
-Backup servos and servo horns 
-Wire and connectors 

-

PaladinEng, 2016-03-31:
is there a backup partial robot?

cdyoder, 2016-03-31:
Do we have an extra bone we can take? And extra acrylic in case the chassis cracks or needs a mod?

closed 2016-06-11 by AhmedSamara