IEEERobotics/bot #473: Slack Vs. IRC

AhmedSamara, 2016-04-22:
During the post-mortem meeting it was pointed out that communication on the team would be a little bit easier if we had a slack or IRC channel for communication like most open-source projects do.

Does anyone have any preferences as to which one we should use?

R00ney, 2016-04-22:
Slack costs money if you what to keep chat history.

mynameis7, 2016-05-03:
I don't think IRC keeps chat history either. I like slack for it's organization through channels, but it's something we could probably achieve through multiple IRCs.

dfarrell07, 2016-05-04:
Both are widely used and would be good experience. In my experience, IRC is used in more open source projects and Slack is typically integrated with other Atlassian tools like JIRA.

Most open source projects that use IRC also use various plugins, like Meetbot for taking notes or logging.

https://meetings.opendaylight.org/opendaylight-meeting/2016/tsc/opendaylight-meeting-tsc.2016-04-21-17.00.html
https://meetings.opendaylight.org/opendaylight-integration/2016/integration/opendaylight-integration-integration.2016-04-28-16.00.html
http://ircbot.wl.linuxfoundation.org/meetings/
http://meetbot.opnfv.org/eavesdrop/%23opnfv-cperf/

AhmedSamara, 2016-06-11:
slack

closed 2016-06-11 by AhmedSamara