IEEERobotics/bot #475: Building the new course (and to see how this ZenHub thing works)

cdyoder, 2016-04-22:
We need to schedule a time where the club can begin fabricating the new course. Can someone create a Doodle Poll/When Is Good so we can figure that out? I know people will be dispersing for exams and summer plans soon, so it may be good to get this out of the way.

PaladinEng, 2016-04-23:
Does this include getting together on a weekend at my place to build? If
so, I believe I can host any weekend after Memorial Day weekend. And just
like last time, I expect the team to have a Cut Plan ready to go prior to
showing up.

SeanKetring, 2016-04-29:
Could everyone meet next Saturday, May 7th to build the course at @PaladinEng 's house?  It is the only day that works for Alwyn's schedule that also falls before I leave for the summer.  

We can use some portion of the old course as building material, and the rest will need to be picked up from Home Depot or someones garage.  We will need to make the BOM by early next week/end of weekend.  

Hope this time works for everyone, we need people to please respond to this thread if you can plan on coming and helping out.

AhmedSamara, 2016-04-29:
May 7 is graduation

SeanKetring, 2016-04-30:
How about May 8th then?

kvijay1995, 2016-05-04:
I can help out this weekend

AhmedSamara, 2016-05-25:
http://doodle.com/poll/ns92nsy64zn8wsse

Here is a google doodle, please respond with what times you're willing to go.

AhmedSamara, 2016-06-11:
Ok so we had to sort through some issues (figuring out how to get our materials out of Troxler), and now have everything in HKN.

However we still have the problem of not having anywhere to store or set up the course.  The new course will have to be collapsable and modular. 

Does anyone know of a good way to do all that?

AhmedSamara, 2016-06-18:
https://drive.google.com/open?id=0B9PYEr43-RtKUXhoV2UwU3hIV1Ehttps://drive.google.com/open?id=0B9PYEr43-RtKUXhoV2UwU3hIV1E

I tried to export the SCAD to get dimensions but can't figure out how to use it.     
But I exported it to other file types to see if any of those are useful. @cdyoder

cdyoder, 2016-08-24:
I get a 404 Error when clicking the link about. Can anyone point me to the model in SCAD or otherwise on the drive?

cdyoder, 2016-09-08:
Just for the record, the SCAD files and the course rules are located [here](https://drive.google.com/drive/u/0/folders/0B93g4TjHMS2pcDFOeERDMEF0SGc) on the drive. NOTE: This is only current as of 9/8/2016. Always check the [Facebook page](https://www.facebook.com/groups/SouthEastCon2017HardwareCompetition/) for updated rules. I'm working on the [SCAD files](http://www.openscad.org/index.html) now for course dimensions. 

In other news, the light saber bit has a lot of electronics inside it (apparently). If there is someone who is itching for an electrical task, this would be good to start!