IEEERobotics/bot #477: Need to move IEEE equipment out of Troxler and into the hallway cabinet.

AhmedSamara, 2016-06-01:
Bobby Compton has just notified me that I've been given back Troxler access for the next week.

We need to move everything out of our own locker, as well as the equipment that's in the closet.

When is a good time for this?

I originally hoped to do this on the Saturday hackday, but I have to be out of town this weekend and can't make it.

I can leave my ID in the locker so everyone can get into troxler without me, just text or email me directly for the password.

jasteve4, 2016-06-01:
Where is the hallway locker?

tabarrett, 2016-06-02:
Hey so are Hack days on Saturday now? I just showed up for the Thursday one, but no ones here. I am about of town this weekend as well but I would be able to help out Sunday afternoon or evening when I get back.

AhmedSamara, 2016-06-03:
The new locker is next to the soda machine.

closed 2016-06-11 by AhmedSamara