IEEERobotics/bot #479: Welcome Back bash preperation.

AhmedSamara, 2016-08-24:
Hello everyone, welcome back bash is this Thursday **(Aug 25) from 4:30-8:00**, we need to get ready for that.    

We're planning for an interest meeting in a classroom reserved for the following **Thursday Sep 1.**

We need:   

- People to [sign up for volunteer hours ](https://docs.google.com/a/ncsu.edu/spreadsheets/d/1ymKfeeG5YuQlyPl7dd2C-LcG2XvNpzCauA9HEJqwxPM/edit?usp=sharing)     (VERY IMPORTANT: I won't be there to set up).

- A flyer with the date for the interest meeting.     (use this https://drive.google.com/open?id=0B_-7IBUxg4smb0FYdFNjVmFYZkdocVZ3WVdSbk9IWlBZMC1F)

- A banner to use for IEEE (I think we can use the same one from packapalooza)

AhmedSamara, 2016-08-24:
@jdekler

cdyoder, 2016-08-24:
Found the old flyer on the drive. New link is here:
https://drive.google.com/open?id=1-KgTyTJE4w9qHiKTHNkCSlJ8vqs27cyW_dDetV35CvI

Thoughts?
-Chris

razcandy, 2016-08-24:
If we want an updated flyer, I can do that tonight.

AhmedSamara, 2016-08-25:
@razcandy did you update the flyer? We also need someone to print it, if anyone is still at home and can print off some copies.

The room is actually EB2-1025, so make sure that's correct.

AhmedSamara, 2016-08-25:
We also need more people to sign up for the table.

Currently only chris is here to help set up, and I have a class on main campus until 4:15.

https://drive.google.com/open?id=19IacQ3nGEE3dyeCd8AqKqQA9u0wPXFfwKNKW0vvj2bQ

Here is the password for the locker (only people with drive access can see it don't worry)

closed 2016-09-01 by SeanKetring